mantidproject/mantid #15876: ISIS sans GUI for new back end

NickDraper, 2016-04-11:
Pre requisiste: #15875

This is to create a new ISIS SANS gui for the refactored back end created in #15875 

Requirements:
1. This should be created seperately from the existing GUI
1. All operations (reduction, centre finding etc) should be able to output the python used to perform the operation (consider "Py" buttons and python notebook creation).
1. Batch processing should be the main mechanism for input, not single file analysis.

gemmaguest, 2018-01-04:
@Mantid-Matthew, I've just noticed that the Python code generated by the Output Notebook option on the new SANS GUI gives this error when you run the notebook: `NameError: name 'SANSGuiDataProcessorAlgorithm' is not defined`.

DavidFair, 2020-04-30:
Closing this as completed, we have a new ISIS SANS GUI written in Python for a while
Any additional issues should be opened separately

closed 2020-04-30 by DavidFair