mantidproject/mantid #15877: Integration assumes that point-data is distribution-data

SimonHeybrock, 2016-04-11:
In `Integration.cpp`:

```cpp
MatrixWorkspace_sptr Integration::getInputWorkspace() {
  // ...
  // To integrate point data it will be converted to histograms
  if (!temp->isHistogramData()) {
    auto alg = this->createChildAlgorithm("ConvertToHistogram");
    // ...
    temp->isDistribution(true);
  }
```

If you convert histogram data (`Y` storing counts) to point-data and then integrate, the result will be wrong.

NickDraper, 2016-04-11:
```python
bob = CreateSampleWorkspace(Function='Quasielastic', UserDefinedFunction='name=LinearBackground,A0=3.22542,A1=-0.0633193')
bobPoint = ConvertToPointData(InputWorkspace='bob')
intPPoint = Integration(InputWorkspace='bobPoint', RangeLower=2000, RangeUpper=4000)

print "this should be 1"
print intPPoint.readY(0)[0]
```

NickDraper, 2016-04-11:
The current code assumes that any point data workspace is a distribution, and forces the flag as such.

In the vast majority of circumstances this would be correct, unless for some reason you have a point data workspace that is not a distribution.

SimonHeybrock, 2016-04-11:
@NickDraper Could you check that, to me it also seems wrong when no range is given (i.e., not only for partial integration).

NickDraper, 2016-04-11:
I'll have a look again.

But I'm not sure anyone should have point data that is not also a distribution.

for example this works
```python
bob = CreateSampleWorkspace(Function='Quasielastic', UserDefinedFunction='name=LinearBackground,A0=3.22542,A1=-0.0633193')
bob = ConvertToDistribution(bob)
bobPoint = ConvertToPointData(InputWorkspace='bob')
intPPoint = Integration(InputWorkspace='bobPoint', RangeLower=2000, RangeUpper=4000)

print "this is 1"
print intPPoint.readY(0)[0]
```

NickDraper, 2016-04-11:
Sorry you are correct about the range based aspect not being important.

However it is true that you should never have a point based dataset that is not a distribution (although I know it is possible to create one in Mantid).  It just would not make much scientific sense.

If you have counts against bin centre values, you did not see 200 counts at 4000ms.  However you might have seen 5 counts/ms at 4000ms, or 200 counts between 4000ms and 4040ms.

Given this I'm not sure this is a very urgent issue.

SimonHeybrock, 2016-04-11:
I think it would be implicitly fixed during the roll-out of the new `Histogram` type.

NickDraper, 2016-04-11:
Then I'll put a low priority label on this and assign it to you to mark as fixed when that is done.

closed 2021-03-03 by stale[bot]