mantidproject/mantid #15886: Quality Plan

OwenArnold, 2016-04-12:
Some sections of the [Quality Plan](https://github.com/mantidproject/documents/blob/master/Project-Management/QP.docx) are out of date. For example reference to `Trac` in section 2.5

Other potential issues:

* Lot of reference to Tessella internal QA documents be inaccessible
* Lots of specific references to Tessella project files that will be inaccessible 
* No mention of developer workflow (I wouldn't expect a lot of detail), just a link to the wiki page.

STFC would like In-kind projects to have quality plans and for **Data Streaming** and **LDV and Instrument Control** we need to cross reference to an accurate Mantid QP.

NickDraper, 2017-03-07:
This is completed in npd/9416

closed 2017-03-07 by NickDraper

OwenArnold, 2017-03-13:
@NickDraper we should talk about this.