mantidproject/mantid #15889: Find better ways to deal with optional Python packages and docs-tests - umbrella

FedeMPouzols, 2016-04-12:
Some of the Python algorithms use Python packages that are not shipped with Mantid.
Sometimes these packages are not available as OS packages for many of our platforms.

Examples of algorithms with issues: 
- `LoadCIF`: requires `PyCifRW` which is not shipped with Mantid, and only available through `pip`.
- `AlignComponents` requires non-ancient scipy which is not available on RHEL6
- Any algorithm that tries to use external applications such as GSAS-II.

Long-term some of these packages might be shipped with Mantid. We need to find a good approach for this for all the supported platforms.

Looking at the docs-test:
- Sphinx needs the algorithms to be registered to be able to create their documentation correctly. If an algorithm fails to do an import at the top level it won't get a proper documentation page.
- We currently have `.. testcode` for examples that run as doc tests, and `.. code-block` for examples that do not run. `code-block` is used sometimes when the doct-test cannot run on all our platforms.

Several improvements were discussed in the last TSC meeting (2016/04/12):

* Algorithms should try to do the risky imports in `PyExec`.
* Add in mantid.api.PythonAlgorithm a try-import method with informative and standard error messages that would be called from the algorithm PyExec methods as the standard way of trying to import non-standard packages.
* It should be possible to run the doc test of an algorithm if the package(s) required for that algorithm is available. Would it be possible to add a different type of block which would be run conditionally on the availability of one or more packages? `.. testcode-if-package` or similar.
* Check where we have `.. code-block` and where we have algorithms that require optional packages or packages not shipped with / not included in the dependencies of Mantid.

- What to do in unit and system tests?

These and related issues could be addressed in various / separate tickets.

MichaelWedel, 2016-04-20:
I've had this problem in `LoadCIF`, there I did the import at some point where the algorithm is actually run. The documentation is done with the `code-block` approach. When the algorithm is run without `PyCifRW` installed it writes a hint to the log.

Unit tests in this case were not problematic because the algorithm logic is fairly well decoupled from the parts that actually depend on the library. Maybe this is something that's feasible at least for similar packages that perform some sort of loading.

For system tests I do not have an answer. There is an open ticket for `LoadCIF` system tests, but there's no way to execute the algorithm if the dependency is not fulfilled.

FedeMPouzols, 2016-05-06:
An example that does a platform specific import at "PyExec" time: BayesQuasi: #15975.

closed 2021-03-01 by stale[bot]