mantidproject/mantid #15941: Results log does not appear in mantid by default

peterfpeterson, 2016-04-19:
On a fresh install (or after deleting all of your qt settings) you no longer have the results log in the application window. It can be added back using the menus, but users don't know to look for it. I don't know when this behavior changed.

NickDraper, 2016-04-20:
I think this was many years ago, The reults log should self appear if you get a warning or higher log message.

Regards,
Nick Draper

NickDraper, 2016-04-20:
#1519 is the ticket that made the results window appear on error to give you an idea about the age of this.

NickDraper, 2016-04-20:
Therefore I suggest we do not change this, as it is not a recent change in behaviour.  However if you strongly disagree we could revisit this @peterfpeterson

closed 2016-05-09 by NickDraper