mantidproject/mantid #15947: Add option on SofQW algorithms to replace NaNs on output

martyngigg, 2016-04-20:
The output from `SofQW` can be used as input for further processing. The presence of NaNs can therefore cause issues in this processing and having the option to replace the NaNs with 0 would allow best of both worlds.

PranavBahuguna, 2016-07-13:
@martyngigg : I've noticed that SofQW is related to several files including SofQW.cpp, SofQWCentre.cpp, SofQWNormalisedPolygon.cpp and so on. Which of these files do I need to actually modify?

martyngigg, 2016-07-13:
@PranavBahuguna There are currently 3 different SofQW algorithms: `SofQWCentre`, `SofQWPolygon` & `SofQWPolygonNormalised`. `SofQW` is what is known as a "workflow algorithm" and simply runs one of the others depending on an input parameter.

For consistency the replacement should be applied on the output of all three so we should try to share code where possible.

PranavBahuguna, 2016-07-14:
In addition for the output, should only y-values be checked or should the x and e-values be checked as well for NaNs?

martyngigg, 2016-07-14:
In theory the error values should only be `nan` when the Y-values have become `nan`. It should be sufficient to then simply replace both `Y` & `E` values when a `NaN` Y value is encountered.

PranavBahuguna, 2016-07-15:
One more question: Are there any particular datasets that cause NaNs to show up in the output workspace?

martyngigg, 2016-07-18:
It's actually the algorithm itself that causes the NaN values to appear. They are used as markers for certain regions of the parameter space that cannot be accessed by the instrument. 

Another group would like to use this as input for other algorithms an need to be able to clean all of the NaNs out of the dataset for it to be usable.

PranavBahuguna, 2016-07-18:
I understand the algorithm does this, but the current input dataset used for testing the SofQW class (IRS26173_ipg.nxs) does not produce any NaNs in the output workspace. In order to test the code, I need to use an input dataset that does produce NaNs in output workspace. Are there any particular datasets that does this?

martyngigg, 2016-07-18:
The current test case for `SofQWNormalizedPolygon` should produce a NaN for the final spectrum in the workspace.

The other 2 algorithms are much less likely to produce a NaN so I would suggesting manually altering the data to include a NaN spectrum before running the test for these cases.

AndreiSavici, 2016-07-18:
See comment in PR:

`Is there a reason why this should not be optional? The correct way is NaN, not 0. ) means that you measured at that point, NaN means that you could not.`

martyngigg, 2016-07-18:
@AndreiSavici This definitely should be optional with the default to be `Off` - it says that in the initial comment. @PranavBahuguna Have you made it optional and added a flag for the user to control it?

PR #16904 (2016-09-12): SofQW replace NaNs in output workspace with zeroes
**Description of work:**

`SofQW` replace NaNs in output workspace with zeroes

**To test:**

Running SofQW algorithm with NormalisedPolygon method with the old code should produce NaNs in the output workspace. The new code replaces any NaNs with zeroes. The test checks all output workspace data for NaNs, and passes if none are found. To test for the existence of NaNs in output, simply run the test.

Fixes #15947.

Release notes have been updated.

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject [... cut]
closed 2016-09-12 by AndreiSavici