mantidproject/mantid #15956: Minor Cleanup of Crystal Field Codes

martyngigg, 2016-04-21:
In [CrystalFieldPeaks.cpp](Framework/CurveFitting/src/Functions/CrystalFieldPeaks.cpp):
* the objects/constants in the anonymous namespace should be treated as statics & their names in all CAPITALS to distinguish them from local variables
* `const bool real = true;`  & `const bool complx = true;` need could have better names to express their intent and again captalizing them
* it would be easier to read if all constants were defined together & then all anonymous functions and not a mixture.

Generally, `boost::lexical_cast` can now be replace with `std::to_string` to remove the boost dependency.

FedeMPouzols, 2016-04-21:
Beware that with `std::to_string` there is apparently no way to control the precision and/or scientific/non-scientific format. So for example tiny values might be printed as 0.000000 which can create a lot of confusion.

martyngigg, 2016-04-21:
Good point, @FedeMPouzols. Isn't that the same with `boost::lexical_cast`, though?

FedeMPouzols, 2016-04-21:
I just double-checked:
```
float fff = 3e-10f;
g_log.information() << boost::lexical_cast<std::string>(fff) << std::endl;
g_log.information() << std::to_string(fff) << std::endl;
```
Will produce:
```
2.99999997e-10
0.000000
```
This reminded me that we got very confused by `to_string` some days ago. It seems that `boost::lexical_cast` implicitly behaves like std streams for floating point numbers. In the boost code I can see `std::stringstream` being used in `lexical_cast/lexical_cast_old.hpp`.

PranavBahuguna, 2016-09-28:
Are we supposed to make changes to CrystalFieldPeaks.cpp or CrystalFieldPeaksBase.cpp? Because all the features you described that you wanted fixing only exist in the latter file.

PranavBahuguna, 2016-09-28:
The object `SYMMETRY_MAP` in the anonymous namespace cannot be declared before the anonymous functions as it uses them in its declaration, so they must be declared first.

PR #17629 (2016-10-03): Minor Cleanup of CrystalFieldPeakBase
Refactor/cleanup of CrystalFieldPeakBase to improve clarity and readability.

**To test:**

Check all the code in [CrystalFieldPeaks.cpp](Framework/CurveFitting/src/Functions/CrystalFieldPeaks.cpp) to ensure that the issues of #15956 have been addressed.

Ensure any tests that utilise CrystalFieldPeakBase pass - these include:
- CrystalFieldMultiSpectrumTest
- CrystalFieldPeaksTest
- CrystalFieldSpectrumTest

Fixes #15956.

*Does not need to be in the release notes.*

#### Reviewer ####

Please comment on the following ([full description](http://www.mantid [... cut]
closed 2016-10-03 by SimonHeybrock