mantidproject/mantid #15960: Displaying a table alters content of string-columns

MichaelWedel, 2016-04-21:
Run the following script in the script editor of MantidPlot:

```
a = CreateEmptyTableWorkspace()

a.addColumn('str', 'col1')
a.addRow({'col1': 'This is readable'})

print a.cell(0, 0)
```

As expected, the output should be:

```
This is readable
```

Display the resulting table workspace. The cell now reads `Thisisreadable`. All spaces have been removed.

I'm adding the High Priority label because this stops `PoldiIndexKnownCompounds` from working should a user ever have the idea to display any of the required input table workspaces.

MichaelWedel, 2016-04-22:
Running in debug mode showed that this is the place where the string is changed:

```
void MantidTable::cellEdited(int row, int col)
```

MichaelWedel, 2016-04-22:
After some more debugging and preserving spaces in the string in the above mentioned function I found that `TableColumn::read` does not work as expected with strings, because it uses `operator<<` to transfer the data from the argument to the stored string. But the operator only proceeds until it encounters a whitespace character, so that changed the text in the column to `This`.

With a template specialisation for `Table<std::string>::read` the detour via the streaming operator can be skipped and these problems go away.

This code has been there for some time, but maybe this `cellEdited`-function was not called before for some reason?

MichaelWedel, 2016-04-22:
The bug does not seem to be in release 3.6, at least on the windows version.

martyngigg, 2016-04-22:
@mantid-roman Could this change have come about from the changes surrounding Qt3?

@MichaelWedel This seems fine to me.

mantid-roman, 2016-04-22:
@martyngigg I am not sure. I can have a look.

martyngigg, 2016-04-22:
@mantid-roman Maybe just see if what @MichaelWedel is suggesting is okay, I think that will be fine.

mantid-roman, 2016-04-22:
The changes look reasonable. I can test the PR when it's ready.

PR #16035 (2016-05-09): Displaying a table no longer removes whitespace characters from string columns
With these modifications string columns are allowed to have whitespace characters again. Does not need to be in the release notes.

**To test:**
Try the example provided in the ticket and make sure that whitespace characters are preserved even after the displaying the table. For columns with numeric types whitespace characters should be removed.

Fixes #15960.

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review ### [... cut]
closed 2016-05-09 by FedeMPouzols