mantidproject/mantid #15981: Vesuvio - VesuvioCommands system test failing on OSX and RHEL6

ElliotAOram, 2016-04-25:
### Expected behavior
* VesuvioCommandsTest should pass on all OS

### Actual behavior
* This system test is failing on OSX and RHEL6 (potentially others)

### Steps to reproduce the behavior
* Run system tests on OSX or RHEL6

### Platforms affected
* OSX, RHEL6

rosswhitfield, 2016-04-25:
I was just looking at this myself, it also fails on Arch Linux.

Here is the diff required so it passes for me: :smile: 
```diff
--- a/Testing/SystemTests/tests/analysis/VesuvioCommandsTest.py
+++ b/Testing/SystemTests/tests/analysis/VesuvioCommandsTest.py
@@ -148,8 +148,8 @@ class SingleSpectrumBackground(stresstesting.MantidStressTest):
 
         _equal_within_tolerance(self, index_one_first, fitted_ws.readY(0)[0])
         _equal_within_tolerance(self, index_one_last, fitted_ws.readY(0)[-1])
-        _equal_within_tolerance(self, index_two_first, fitted_ws.readY(1)[0])
-        _equal_within_tolerance(self, index_two_last, fitted_ws.readY(1)[-1])
+        _equal_within_tolerance(self, index_two_first, fitted_ws.readY(1)[0], tolerance=0.13)
+        _equal_within_tolerance(self, index_two_last, fitted_ws.readY(1)[-1], tolerance=0.11)
 
         fitted_params = self._fit_results[1]
         self.assertTrue(isinstance(fitted_params, MatrixWorkspace))
```

But this should be tested on OSX and RHEL6

ElliotAOram, 2016-04-25:
Thanks for looking at that @rosswhitfield much appreciated. Do you know if it is only the `SingleSpectrumBackground` test that is failing in VesuvioCommandsTest?
I'm hoping to find out what the difference is being caused by but I feel like it might be down to fitting

rosswhitfield, 2016-04-25:
Yes just `SingleSpectrumBackground`

ElliotAOram, 2016-04-26:
Test changes pass locally on:
* [ ] Windows
* [ ] RHEL6
* [ ] Ubuntu

PR #15988 (2016-05-04): Vesuvio - VesuvioCommands check height of peak rather than first/last bin
The VesuvioCommandsTest now checks the maximum height of the calculated data for comparison. This value was seen to be far more consistent across different platforms rather than the first/last bin which were seemingly sensitive to the noisy data at the start and end 

**To test:**

* Ensure the system test `VesuvioCommandsTest` passes ideally checking this on a few different OS. 
  * I have checked this locally on Windows, Centos 6 and Ubuntu 14.04 and it passes

Fixes #15981

*Does not need to be in the release notes.*
 [... cut]
closed 2016-05-04 by martyngigg