mantidproject/mantid #15989: Indirect ISIS Calibration - Load Log Files option in interface

ElliotAOram, 2016-04-26:
### Expected behavior
* ISIS Calibraction interface *(Interfaces > Indirect > Data Reduction > ISIS Calibration)* should allow for the **optional** loading of log files

### Actual behavior
* You are forced to load the log files along with the data

### Steps to reproduce the behavior
* Add a single raw file (and no log files) to a directory
* Make this the only directory available to mantid 
* Attempt to Load this file using ISIS Calibration interface

### Platforms affected
* All

SpencerHowells, 2016-04-26:
The default value should be kept True otherwise users will complain that ElWin does not find the temperature. This is the reason why it is fixed True!

SpencerHowells, 2016-04-26:
We also have Issue #13746 regarding Loading of log files

SpencerHowells, 2016-07-05:
Just realised that this refers to Calibration which does not need log files. Default should be False.

ElliotAOram, 2016-07-14:
# ToDo:

* [ ] Add check box widget to ISIS Calibration ui (`ISISCalibration.ui`) 
  * see `ISISEnergyTransfer.ui` for example
* [ ] Add an additional property to the algorithm `IndirectCalibration.py` to allow a boolean option for `LoadLogFiles` and ensure `Load` uses this value in the interface. Default should be false.
* [ ] Get the state of the checkbox widget in `ISISCalibration.cpp` `run()` method
* [ ] Add this property to the `ISISCalibration.cpp` `run()` method
* [ ] Handle the error nicely if log files can't be found in the directories

# To test
* add a single raw file to a folder
* make sure this is the only folder mantid can see
* try to load with out logs (should pass) 
* try to load with logs (should fail and handle error nicely)
* add the matching log file to the folder
* Load with (should pass)
* Load without (should pass - ensure that the workspace does not have log info)

PR #16940 (2016-07-22): Indirect Reduction ISIS Calibration - add load log option
Added a load log option to indirect calibration interface.

**To test:**

Load a data file with logs in repository. Run with load log option checked and then unchecked and compare results in sample logs. Repeat without logs in repository. 

Fixes #15989

[Release Notes](https://github.com/mantidproject/mantid/blob/e6b65d7f553430e77b974cdd305a16d23cc808bf/docs/source/release/v3.8.0/indirect_inelastic.rst)

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)) [... cut]
closed 2016-07-22 by FedeMPouzols