mantidproject/mantid #16007: Indirect Elwin - Ungroup Elwin Output

ElliotAOram, 2016-04-27:
### Expected behavior
* The output from Elwin should not be grouped

### Actual behavior
* The output of the algorithm is added to a group

### Steps to reproduce the behavior
* Run the Elwin interface with any input files

### Platforms affected
* All

SpencerHowells, 2016-04-29:
Perhaps this should be an option. When used in a temperature scan, the input could be dozens of runs. If ungrouped, it would really clutter up the Workspace pane with entries.

ElliotAOram, 2016-04-29:
@SpencerHowells, good idea I'll implement as an option instead

ElliotAOram, 2016-05-16:
This is now working but the branch will have to be merged with master and the release notes updated when they are available for 3.8 due to the code freeze.

PR #16640 (2016-06-20): Indirect Data Analysis - Ungroup Elwin output option
Add an option in the Elwin interface to ungroup the input files. Files should be group by default

**To test:**

* Open the Elwin interface (Interfaces > Indirect > Data Analysis > Elwin)
* Load in the input File `irs26176_graphite002_red.nxs` (available from unit test data directory)
* Ensure the `Group Input` option is checked
* Click `Run`
  * Ensure a workspace named `IDA_Elwin_Input` is created and contains the input file.
* Uncheck the `Group Input` option 
  * Ensure no group workspace is created (or it is removed if it was already th [... cut]
closed 2016-06-20 by FedeMPouzols