mantidproject/mantid #16039: Vesuvio - Intensity constraints not functioning as expected

ElliotAOram, 2016-05-03:
### Expected behavior
* The intensity of the peaks in the Vesuvio fitting should be as described in the `[intensity_constraints']` flag in the driver script

### Actual behavior
* The second fit in VesuvioTOFFit that is used to obtain the Chi squared value appears to change the values that should be constrained 

### Steps to reproduce the behavior
* Run a complex intensity constrained vesuvio fit e.g.
`flags['intensity_constraints'] = list([2.7593,-0.6864,0,0,0]) , list([0,0,0,2.1580,-0.4968]) ,  list([0,0.4968,0,-2.7593,0])`

### Platforms affected
* All

martyngigg, 2016-05-13:
@ElliotAOram I've had to revert the merge to master as all of the clean builds broke. We don't yet understand why the pull request builds didn't pick this up.

If you download [vesuvio_intensity_patch.txt](https://github.com/mantidproject/mantid/files/262672/vesuvio_intensity_patch.txt) and switch to the top of the mantid source on your machine you can get the changes back onto a new branch by running:

````
git co --no-track -b 16039_vesuvio_intensity_constraints_fixed
git apply vesuvio_intensity_patch.txt
````

ElliotAOram, 2016-05-20:
This issue will now close #15600

ElliotAOram, 2016-05-20:
All Vesuvio tests passed locally on windows 7 and Ubuntu 14.04

ElliotAOram, 2016-05-27:
Merged with release branch for 3.7 hence can be closed

PR #16060 (2016-05-12): Vesuvio - VesuvioTOFFit incorrect params output
VesuvioTOFFit now returns the parameters from the initial Fit rather than those of the secondary fit that is used to calculate Chi Squared.

**To test:**

**I'd recommend building this in Release mode. The Vesuvio Algorithm, can run very slowly in Debug**

* Run the `Vesuvio_intensity_constraints_test.py` script available from `\\olympic\babylon5\Public\ElliotOram\Vesuvio` in the mantidplot scripting window with the DataArchive searching enabled.
  * You will also need the `IP_2015_06_Chimney_header.txt` in a directory where mantid can find it (t [... cut]
PR #16334 (2016-05-24): vesuvio intensity constraints fixed
VesuvioTOFFit now returns the parameters from the initial Fit rather than those of the secondary fit that is used to calculate Chi Squared.

**Further updates to this have been made to ensure that all Vesuvio tests pass. All tests have passed locally on Ubuntu 14.04 and Windows7**

**To test:**

**I'd recommend building this in Release mode. The Vesuvio Algorithm, can run very slowly in Debug**

* Run the `Vesuvio_intensity_constraints_test.py` script available from `\\olympic\babylon5\Public\ElliotOram\Vesuvio` in the mantidplot scripting window with the Da [... cut]
closed 2016-05-27 by ElliotAOram