mantidproject/mantid #16045: Add docs-test example for LoadNexus with LoadTOFRawNexus

FedeMPouzols, 2016-05-03:
See #15959.
LoadNexus can load 4 distinct types of nexus files using other, more specific, algorithms:
- LoadMuonNexus
- LoadNexusProcessed
- LoadISISNexus
- LoadTOFRawNexus

For the three first types there is an example in the documentation of `LoadNexus`: http://docs.mantidproject.org/nightly/algorithms/LoadNexus-v1.html. The fourth type is missing an example (which would also link to LoadTOFRawNexus). There is a small unit test file that could be used for this (see #15959).

KarlPalmen, 2016-06-29:
LoadTOFRawNexus does not have a usage example. See http://docs.mantidproject.org/nightly/algorithms/LoadTOFRawNexus-v1.html .

KarlPalmen, 2016-06-29:
Issue #14595 deals with the usage example for LoadTOFRawNexus. In it @peterfpeterson asks about whether this algorithm is still used.

FedeMPouzols, 2016-08-10:
I'd say this can be closed as a duplicate of #14595. We need to figure out if the algorithm is in use and if so, find a good instance for a usage example / doc-test.

closed 2017-01-16 by NickDraper