mantidproject/mantid #16067: Investigate SpecularReflectionCalculateTheta results for RROA

MattKing06, 2016-05-05:
### Expected behavior

- An appropriate value for ThetaIn should be calculated usually 0 < thetaIn <= 3.5

### Actual behavior

- A value of around -1.0e-05 is calculated when using an INTER dataset

### Steps to reproduce the behavior

- Open `Interfaces > Reflectometry > ISIS Reflectometry (Polref)`
- Choose processing Instrument as INTER
- Enter Run number `13463`
- Process
- See that the algorithm fails and a value of -1e-05 is populated in the theta column.

### Platforms affected
- All

raquelalvarezbanos, 2016-05-05:
Something I noticed (I think it is related to the interface rather than `SpecularReflectionCalculateTheta`:

When running the following reduction (on master branch):

[image]

(note that the first run is INTER and the second one POLREF) the result is:

[image]

- The reduction fails on INTER because the calculated incident theta is negative. If I set `AnalysisMode=MultiDetectorAnalysis` the reduction fails because spectrum 6 can not be found (input workspaces has 5 spectra)
- The reduction on POLREF fails because the calculated `Qmin` and `Qmax` are both zero for some reason (maybe because it assumes `ThetaIn=0`?) If I set those values to `0.1` and `0.3` for instance, the reduction is successful but the `Angle` column is not populated for some reason.

We have to understand
- Why the calculated angle is negative for INTER
- Why the `Angle` column is populated for INTER but not for POLREF
- And if `Qmin` and `Qmax` are both zero for POLREF because `ThetaIn` was not populated in the table.

MattKing06, 2016-05-19:
After some investigation I have found the explanation to some of these issues we are seeing:

- The negative angle for INTER appears due to the `autoFillRow()` method that was in `ReflTableViewPresenter` which would search the Logs for a `twoTheta` tag and retrieve that value to populate the table. The reason the angle was negative here was due to the fact that run 13463 for INTER was a transmission run and thus the angle was not a relevant measurement.

- We weren't sending the value of theta produced by ReflectometryReductionOneAuto back to the presenter to fill the row. The only part of the code that was filling the row was again `autoFillRow()`. There was no `twoTheta` value in the POLREF logs and thus no value could be found to populate the table.

- In ReflTableViewPresenter theta was set to zero and then the value was updated if it could be found in the logs or if the column had been filled in for that row. POLREF had no theta value in the logs and the theta column for that row was empty which meant `theta` kept its value of zero. This value was then used to calculate Qmin and Qmax using `sin(0)` which result in both Qmin and Qmax being zero.

I suspect any of the problems found with SpecularReflectionCalculateTheta were actually due to incorrect use of the algorithm by ReflectometryReductionOne meaning that this issue can be closed unless we find another issue with this algorithm.

closed 2016-05-19 by MattKing06