mantidproject/mantid #16068: QxQz slicing issue

raquelalvarezbanos, 2016-05-05:
Jos reported the following issue:

They have the following QxQz workspace:

[37513_qxqz.zip](https://github.com/mantidproject/mantid/files/250377/37513_qxqz.zip)

which looks like this on Slice Viewer:

[image]

They have their own integration script:

[image]

That produces a similar cut in shape but with different magnitude:

[image]

They have noticed that there is a Rebin2D step they were not expecting that could be the cause of this difference.

This issue is to clarify what Slice Viewer is doing differently from their script.

gemmaguest, 2020-07-17:
The difference is that `Integration` is doing a sum of Y values * bin width because this is distribution data. Slice viewer uses `Rebin2D` which just sums Y values.
```
ws=Load(Filename='37513_qxqz.nxs', OutputWorkspace='37513_qxqz')

def QxQzcuts(name, qzmin=None, qzmax=None, outputname=None):
    if not outputname:
        if qzmin:
            outputname = name+'_cut_'+str(qzmin)+'-'+str(qzmax)
        else:
            outputname = name+'_cut_all'
    Transpose(InputWorkspace=name, OutputWorkspace=outputname)
    Integration(InputWorkspace=outputname, RangeLower=qzmin, RangeUpper=qzmax, OutputWorkspace=outputname)
    return Transpose(InputWorkspace=outputname, OutputWorkspace=outputname)

# The input is distribution data so Integration() sums Y values multiplied by bin width
QxQzcuts('37513_qxqz',0.03,0.05, '37513_qxqz_Integration')

# If we set it as non distribution it will just sum Y values - gives more similar results to Rebin2D below
ws.setDistribution(False)
wsInt=QxQzcuts('37513_qxqz',0.03,0.05, '37513_qxqz_Integration_non_dist')
ws.setDistribution(True)
wsInt.setDistribution(True)

# The sliceviewer creates the cut with Rebin2D as follows:
Rebin2D(InputWorkspace='37513_qxqz', OutputWorkspace='37513_qxqz_Rebin2D',
  Axis1Binning='-0.0005, 0.000002, 0.0005', Axis2Binning='0.03,0.02,0.05', UseFractionalArea=True)
```

gemmaguest, 2020-08-06:
Closing as Stale because this is expected behaviour at the moment

closed 2020-08-06 by gemmaguest