mantidproject/mantid #16069: Can't fit data from table with fit property browser

tom-perkins, 2016-05-05:
### Expected behaviour
Should be able to use the "Fit function" window in MantidPlot to fit data from a TableWorkspace

### Actual behaviour
Dialog "Fit algorithm failed. Unknown property search object WorkspaceIndex."  
(More recently this has changed into an error message in the log)

### Steps to reproduce the behaviour
```python
tab = CreateEmptyTableWorkspace()
tab.addColumn("int", "x", 1)  
tab.addColumn("double", "y", 2)         
tab.addRow([20, 40])
tab.addRow([30, 60])
tabl = importTableWorkspace('tab', True)
```
Then plot a graph from the table and try to use the fit property browser ("Fit Function" window in MantidPlot) to fit something to it. 

### Platforms affected
All

---

This looks like it occurs in `doFit` in the FitPropertyBrowser. The function calls `getWorkspace()` which automatically converts the data to a MatrixWorkspace, but then does nothing with the workspace returned. It just passes the name of the original TableWorkspace to `Fit`.

**This is required soon for a course at ISIS - deadline 14th May**. (Use case: the results table generated from MuonAnalysis interface. The workaround is manually converting to a MatrixWorkspace).

shahrozahmed1, 2016-05-09:
I built the branch again, it seems to crash whenever I click `peak fit toolbar` after plotting the graph.

[image]

shahrozahmed1, 2016-05-09:
@tom-perkins its quite odd because now I am not getting crash any more, only happens randomly..

PR #16072 (2016-05-10): Support fitting data from table workspaces
**[Required for a course at ISIS on 14th May]**

Fix a bug where attempting to fit data plotted from a TableWorkspace failed.

The FitPropertyBrowser has a method called `getWorkspace()` which converts a TableWorkspace to a MatrixWorkspace to use internally. The result of this method was never used.
Changed this so that the MatrixWorkspace is passed to the `Fit` algorithm.

Checked that the correct name still appears in the workspace history when fitting a MatrixWorkspace as usual.
(If fitting a TableWorkspace, a 'temporary workspace' name will appear [... cut]
closed 2016-05-10 by FedeMPouzols