mantidproject/mantid #16074: Tabulated version of StretechedExFT fit function

jmborr, 2016-05-05:
The current StretechedExFT employs scipy.fftpack.fft each time the function is evaluated. As a consequence, it is slow and carries all the problems and glitches of the fast fourier transform. Here we vill implement a tabulated version. Tables of the Fourier transform of the stretched exponential function have been carried out for an extensive energy domain and range of beta values.

SpencerHowells, 2016-05-06:
Will this be a change or an option? From a science point of view I'd like to be able to see if the 2 methods give the same result. I agree that the fft method takes too long for general use. With the tabulated version I could then use the Fabada Bayesian fitting routines.

jmborr, 2016-05-06:
Curve fitting functions in Mantid do not have versions like the algorithms, so it will be a replacement. However, we can keep the FFT implementation by adding attribute "mode" to the function. This attribute  could take values "tabulated" (the default) and "FFT", and would determine whether to use the tables of FFT to evaluate the function. The tables are faster but they are good only in the range of stretching exponents and energies for which they are defined. There may be the odd case when one would like to fit to very high stretching exponents (>3) and energies (~meV)

SpencerHowells, 2016-05-09:
Having a 'mode' attribute was what I was thinking of. Regarding the stretching exponent (beta), from the science it must be in the range 0 to 1. In my fitting I have that as a constraint.

jmborr, 2016-05-24:
Issue superseeded by #16397. The increase in speed makes a tabulated version unnecessary.

closed 2016-05-24 by jmborr