mantidproject/mantid #16077: Error propagation incorrect in DivideMD, MultiplyMD

AndreiSavici, 2016-05-05:
### Expected behavior
I have two MD histogram workspaces, `a` and `b`, and I want to look at `a/b`
If I have a point in my LHSWorkspace where the signal and the error squared are 0, I expect that the error squared is 0. 

### Actual behavior
In the current implementation I get a `NaN`

### Steps to reproduce the behavior
```
d=CreateMDHistoWorkspace(SignalInput='0', ErrorInput='0', Dimensionality=1, Extents='0,1', NumberOfBins='1', Names='d1', Units='x')
n=CreateMDHistoWorkspace(SignalInput='1', ErrorInput='0', Dimensionality=1, Extents='0,1', NumberOfBins='1', Names='d1', Units='x')
out=a/b
print out.getErrorSquaredArray()
```
### Platforms affected
All

### Notes:
 - The same behavior was fixed  for `Divide` algorithm in 2010. See comments in Divide.cpp, lines 47-50
 - MultiplyMD has the same problem

NickDraper, 2016-05-20:
Moving this to R3.8

AndreiSavici, 2016-05-20:
Moving it back to 3.7. Error propagation is important enough to have it done correctly

PR #16335 (2016-05-25): Avoid division by 0 in MD error propagation
Explicitly perform operations to avoid division by 0. 

**To test:**

Make sure all unit test, system tests, and doc tests that use multiplication and division of MD workspaces are OK.

The following script should yield `[0.]`. The same if you replace division by multiplication. (The current version returns NaN)
 
```
d=CreateMDHistoWorkspace(SignalInput='0', ErrorInput='0', Dimensionality=1, Extents='0,1', NumberOfBins='1', Names='d1', Units='x')
n=CreateMDHistoWorkspace(SignalInput='1', ErrorInput='0', Dimensionality=1, Extents='0,1', NumberOfBins= [... cut]
closed 2016-05-25 by martyngigg