mantidproject/mantid #16082: Crash on click column option in MantidPlot table

martyngigg, 2016-05-06:
### Expected Behaviour

No crash

### Actual behavior

Crash

### Steps to reproduce the behavior

Run the following script:

```python
w = CreateEmptyTableWorkspace()
w.addColumn('str', 'Code')
w.addColumn('str', 'Value')
w.addRow(['',''])
w.addRow(['',''])
```

Double click on the table to view it, right click on the *Code* column and click column options and  :boom: 

### Platforms affected

All

martyngigg, 2016-05-06:
@tom-perkins This mimics what is done with James Lord's `CreateEmptyTableForQuantum` algorithm that a user just showed me. `v3.6.0` does not crash so we've introduced the bug this release. I suspect something around the Qt3 changes so @mantid-roman may be able to help.

AndreiSavici, 2016-05-06:
This is duplicate of #15915

martyngigg, 2016-05-06:
Confirmed as duplicate of #15915. I'll add to the PR  as fixing that issue too.

PR #16084 (2016-05-06): Fix crash on clicking Column Options for a table
Check for null pointer before attempting to delete it.

[A call to freeMemory()](https://github.com/mantidproject/mantid/blob/c8db225f7cea3f33eff0b464133d6b2633111e43/MantidPlot/src/Table.cpp#L1601-L1602) was recently added when a Table is saved
to memory. Add a check in this function to avoid attempting to
free memory that has already been freed.

**To test:**

- Run the example script below (from issue). 
```python
w = CreateEmptyTableWorkspace()
w.addColumn('str', 'Code')
w.addColumn('str', 'Value')
w.addRow(['',''])
w.addRow(['',''])
tab = i [... cut]
closed 2016-05-06 by AndreiSavici