mantidproject/mantid #16119: TOFSANSResolution does not deal with uncertainty of the last bin edge?

SimonHeybrock, 2016-05-10:
In `TOFSANSResolution.cpp` line 114 we have:
```cpp 
  // Initialize Dq
  MantidVec &DxOut = iqWS->dataDx(0);
  for (int i = 0; i < xLength - 1; i++)
    DxOut[i] = 0.0;
```

For some reason the algorithm does not deal with the `Dx` of the last bin edge, and I don't understand if there is a reason for this, or if it is an oversight. This continues throughout the exec method of that algorithm.

@mdoucet Have a look at this, this might be relevant for you.

mdoucet, 2016-05-10:
I think that's an oversight.

SimonHeybrock, 2016-05-13:
@mdoucet We are considering changing they way x-uncertainties are handled under the hood. As far as I understand so far, `Dx` is actually the Q-resolution for a certain bin, so should actually be a vector of the same length as `Y`.

Is this correct also for all of your applications?

I believe this change can be made without breaking much, and would tidy up some mess in code that deals with that extra (unwanted) value at the end of the `Dx` vector we have currently. In the end, this should result in code that is easier to read, understand, and modify.

mdoucet, 2016-05-13:
@SimonHeybrock Right. That's why we ended up having to juggle things around to deal with the last bin boundary. Having a way to track dQ and keep it in the workspace after any other algorithm is called on the workspace would be a major improvement.

PR #16141 (2016-05-17): Assign resolution properly on bin boundaries
This PR deals with two issues:
 1- The Q resolution values were not assigned properly, according to #16119.
     I changed my mind on this one. Playing around with the values at bin boundaries creates artifacts in the Dx array. It's preferable to leave it as it is so that both the workspace and the saved I(q) file can safely be used. The original problem only affects the last point of the distribution, only when you move to point data. That point has a dQ that is underestimated. It's best to deal with this in the refactor mentioned in #16119.
 2- Th [... cut]
closed 2016-05-17 by martyngigg