mantidproject/mantid #16120: SaveAscii2 assumes that workspaces without common bins store histogram data

SimonHeybrock, 2016-05-10:
I am not aware of any issues seen in the wild, but this looks suspicious:

`SaveAscii2.cpp` around line 300:
```cpp
    if (!m_isCommonBins) // checking for ragged workspace
    {
      file << (m_ws->readX(*spectraItr)[bin] +
               m_ws->readX(*spectraItr)[bin + 1]) /
                  2;
    }
```
In other cases (i.e., for common bins), there is a check for `isHistogramData()`, but this seems to be missing here.

SimonHeybrock, 2016-06-22:
I leave this unassigned since it is not clear that it will be me who will be working on introducing the Histogram type in this context (#15880).

NickDraper, 2016-06-22:
@SimonHeybrock As you have #15880 you might as well hold this one as well.  It is better than leaving it unassigned as it is likely to get missed there, and gets in the way of me triaging other issues.

SimonHeybrock, 2017-02-17:
It seems this has been fixed automatically by the `HistogramData` rollout a while ago.

closed 2017-02-17 by SimonHeybrock