mantidproject/mantid #16146: Mantid crashes during saving of project

mganeva, 2016-05-11:
Operation with groups of workspaces lead to very long names of the resulting workspaces. It causes mantid crash during saving of project.

### Expected behavior
Get the Mantid project saved.

### Actual behavior
Mantid crash

### Steps to reproduce the behavior
Create a workspace with the very long name, for example:
``` python
wsname = 'bh28000019_van_ar27630120leer_sfleer_scaled_bh28_Van-300K_with-all-corr_sfdata_group_bh28_Van-300K_with-all-corr_vana_coefs_2theta-11.56_bh28_Van-300K_with-all-corr_sfdata_group_vanacorr_bh28000020_van_ar27630121leer_nsfleer_scaled_bh28_Van-300K_with-all-corr_nsfdata_group_bh28_Van-300K_with-all-corr_vana_coefs_2theta-11.56_bh28_Van-300K_with-all-corr_nsfdata_group_vanacorr_ar27630060nicr_ar27630061nicr_ar27630120leer_sfleer_scaled_ar27630121leer_nsfleer_scaled_bh28_Van-300K_with-all-corr_nsf_data_nicrcorr'
CreateSampleWorkspace(OutputWorkspace=wsname)
plotSpectrum(wsname, 1, True)
```
Click Menu File->Save Project As..  to save the project.

### Platforms affected
I've observed this crash on mac os x (mavericks and el capitan). I have not tested it on other platforms.

samueljackson92, 2016-06-29:
As far as I can see this is actually nothing to do with project loading/saving. The crash is due to Mantid falling over in NexusFileIO, specifically on an ```NXopen``` call. I can reproduce this without attempting to save the project using the following script:

```python
wsname = 'bh28000019_van_ar27630120leer_sfleer_scaled_bh28_Van-300K_with-all-corr_sfdata_group_bh28_Van-300K_with-all-corr_vana_coefs_2theta-11.56_bh28_Van-300K_with-all-corr_sfdata_group_vanacorr_bh28000020_van_ar27630121leer_nsfleer_scaled_bh28_Van-300K_with-all-corr_nsfdata_group_bh28_Van-300K_with-all-corr_vana_coefs_2theta-11.56_bh28_Van-300K_with-all-corr_nsfdata_group_vanacorr_ar27630060nicr_ar27630061nicr_ar27630120leer_sfleer_scaled_ar27630121leer_nsfleer_scaled_bh28_Van-300K_with-all-corr_nsf_data_nicrcorr'
CreateSampleWorkspace(OutputWorkspace=wsname)
plotSpectrum(wsname, 1, True)
SaveNexus(wsname, wsname)
```

My guess is that the filename is too big to be supported by the filesystem? Stack overflow tells me the default Mac filesystem only supports 255 characters and that filename is 500+ so it could be that? Shall look into it more soon. Obviously, whatever the cause we can do better than just crashing...

mganeva, 2016-06-30:
@samueljackson92 I am not sure that it is a filesystem problem. If I do right click on the workspace and choose SaveNexus, it works. The crash (on both, mac and debian) happens only during saving of project.

In any case, the correct behaviour would be not to crash, but to inform user that the workspace name is too long and could not be saved.

samueljackson92, 2016-07-01:
@mganeva Huh. That's weird. When I run your script and do a right click > SaveNexus Mantid crashes for me (OSX El Capitan). What happens if you run the script in my comment above? Also (just in case) what version of Mantid are you observing this in? I get the right click and script crash on both the 3.7.1 release and on a recent master build.

Yes I agree a message would be the most logical solution. It should be fairly trivial to put something in the ApplicationWindow code where the SaveNexus/Save project calls are being made.

mganeva, 2016-07-01:
@samueljackson92 I have updated version of Mantid (to release 3.7.1). Now it crashes with the message: "Application Specific Information: detected buffer overflow" if I run your script (OSX El Capitan). I don't know why it did not crash in earlier version. Since it was not asking me for a file name, I suppose that it was taking some default name which was short. 

On debian jessie it also crashes with "buffer overflow detected : MantidPlot terminated" message. So, it is not only the mac os x problem.

It looks like a filesystem problem. If I do 
```
$ touch bh28000019_van_ar27630120leer_sfleer_scaled_bh28_Van-300K_with-all-corr_sfdata_group_bh28_Van-300K_with-all-corr_vana_coefs_2theta-11.56_bh28_Van-300K_with-all-corr_sfdata_group_vanacorr_bh28000020_van_ar27630121leer_nsfleer_scaled_bh28_Van-300K_with-all-corr_nsfdata_group_bh28_Van-300K_with-all-corr_vana_coefs_2theta-11.56_bh28_Van-300K_with-all-corr_nsfdata_group_vanacorr_ar27630060nicr_ar27630061nicr_ar27630120leer_sfleer_scaled_ar27630121leer_nsfleer_scaled_bh28_Van-300K_with-all-corr_nsf_data_nicrcorr
```
I get a message that "File name too long" on both systems.

samueljackson92, 2016-07-01:
@mganeva That makes sense. I've played around some more this afternoon. I wrote a small pure C++ program that tries to write out files with an incrementally larger file name and it fails to create a file as soon as the filename hits 256 characters on OS X.

As no error code is being returned from the NXOpen call in SaveNexus something must be doing something naughty in either the Nexus Format or HDF5 libraries, which are beyond our control to fix.

In the short term it would be easiest to simply throw an exception here and let the algorithm runners handle it. That way it should always be caught and the user gets notified. I've just written that fix and I'll throw it up to the build shortly.

In the long term it might be an idea to try and doing something more sensible about how those long file names get generated. Any idea what algorithms were run in order for that particular workspace name to be generated?

quantumsteve, 2016-07-01:
@samueljackson92 There is some overlap between Mantid and NeXus developers. A self-contained repro or a pull request addressing the issue should be sent upstream.

https://github.com/nexusformat/code

samueljackson92, 2016-07-04:
@quantumsteve Yes you're absolutely right. I was aware, but I only thought to report it upstream after I was driving home Friday 😄

closed 2016-07-04 by quantumsteve

mganeva, 2016-07-05:
@samueljackson92 That particular workspace name has been generated during applying of different algorithms (standard DNS data reduction workflow) to the groups of workspaces. Operation on the groups of workspaces is very convenient and we use it a lot.

Simple script which shows how the name of the workspace growth:
```
# create 2 groups of sample workspaces
for i in range(4):
    wname = 'ws' + str(i)
    CreateSampleWorkspace(OutputWorkspace=wname)
    
group1 = GroupWorkspaces('ws0,ws1')
group2 = GroupWorkspaces('ws2,ws3')

# do some operation, save the result into a new group
sum = 'group1'
for i in range(10):
    wname = 'sum_'+str(i)
    Plus(sum, group2, OutputWorkspace=wname)
    sum = wname
```

samueljackson92, 2016-07-05:
@mganeva Yes I can see how that would easily occur. If there are particular workflow algorithms that are generating excessively long names we could modify them to call their output to something shorter. However I appreciate that applying multiple independent steps can still produce similar results, which is something that we can't really do too much about.

PR #16774 (2016-07-04): SaveNexus crashes with long file name
#16146 reported a crash when file names are extremely long. A little digging shows that this is due to something going wrong in the NXopen call in SaveNexus. As we can't directly edit that code here's a fix that throws an exception in the event that the file name is too long and the file cannot be created. 

**To test:**

Here's a script that should reproduce the error. The first file should be saved correctly, but Mantid should crash on saving the second file. Applying the fix should show that the first file still saves correctly but that the second file  [... cut]