mantidproject/mantid #16151: Engg GUI: Vanadium curves not being plotted for north or south bank

shahrozahmed1, 2016-05-12:
### Expected behavior
After the completion of calibration, there should be vanadium curves being plotted for each bank separately. 

### Actual behavior
Currently curves are being plotted for one of the bank, it is possible that `Bank 1` is being plotted which is then overwritten with `Bank 2`. 

### Steps to reproduce the behavior

Run `Calibration` on `Engineering Diffraction Analysis` GUI

### Platforms affected
All.

shahrozahmed1, 2016-05-12:
@FedeMPouzols It seems that `EnggVanadiumCorrections` seems to have just the one output for curves [here](https://github.com/mantidproject/mantid/blob/master/Framework/PythonInterface/plugins/algorithms/EnggVanadiumCorrections.py#L39-L41). Does that mean this curves is for both the banks?

FedeMPouzols, 2016-05-12:
That produces one workspace, but that one has the curves for as many banks as you calculate. For each bank there should be the three curves.
Follow to https://github.com/mantidproject/mantid/blob/master/Framework/PythonInterface/plugins/algorithms/EnggVanadiumCorrections.py#L103
Then 
https://github.com/mantidproject/mantid/blob/master/Framework/PythonInterface/plugins/algorithms/EnggVanadiumCorrections.py#L162
And then "_fitCurvesPerBank":
https://github.com/mantidproject/mantid/blob/master/Framework/PythonInterface/plugins/algorithms/EnggVanadiumCorrections.py#L213

FedeMPouzols, 2016-05-12:
It's explained in the documentation of the property: https://github.com/mantidproject/mantid/blob/master/Framework/PythonInterface/plugins/algorithms/EnggVanadiumCorrections.py#L61-L64

shahrozahmed1, 2016-05-12:
@FedeMPouzols yes you are right but it seems on the GUI, just tested it, the problem is `Force recalculate and overwrite all existing intermediate files` check-box doesnt exactly work for some odd reason, therefore we may not have utilised `EnggVanadiumCorrections` since that broke?

PR #16229 (2016-05-17): EnggGUI:: Vanadium curves now plotting for both banks after calibration
**Description of work.**

After calibration has completed on the Engineering Diffraction GUI, the vanadium curves should now be plotted for each bank separately. 

**To test:**

Go to: `Interfaces/Diffraction/Engineering Diffraction/Settings`
Go to the `Calibration` directory listed

Either:
- Delete the existing files which would only generate vanadium curves for one bank
- Change the calibration directory 

Files to delete should look something like this: `ENGINX_precalculated_vanadium_run00236516_bank_curves.nxs` & `ENG [... cut]
closed 2016-05-17 by martyngigg