mantidproject/mantid #16163: Unscripted Testing ISIS Indirect interfaces

NickDraper, 2016-05-12:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this ticket once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

See
* http://www.mantidproject.org/Indirect:ConvertToEnergy
* http://www.mantidproject.org/Indirect:Indirect_Data_Analysis
* http://www.mantidproject.org/UsingElwin
* http://www.mantidproject.org/UsingFury

ElliotAOram, 2016-05-16:
In depth testing on Windows 7 and more general checks on Ubuntu 14.04

ElliotAOram, 2016-05-17:
The following new issues have been found in Indirect Unscripted testing:
* [ ] #16245 
* [ ] #16247 
* [ ] #16248 
* [ ] #16251 
* [ ] #16255 
* [ ] #16258 
* [ ] #16259 
* [ ] #16261

louisemccann, 2016-09-23:
The following issues have been found in Indirect Unscripted testing:

- [ ] #17543
- [ ] #17560
- [ ] #17573

closed 2017-10-03 by NickDraper