mantidproject/mantid #16166: Unscripted Testing SCD event data reduction

NickDraper, 2016-05-12:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this ticket once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

lottiegreenwood, 2016-05-19:
windows 7 64bit
tested using instructions [here](http://docs.mantidproject.org/nightly/interfaces/SCD_Event_Data_Reduction_Interface.html) no major issues found

lottiegreenwood, 2016-09-23:
windows7, tested as above. no obvious issues found

NickDraper, 2017-02-09:
not done but too late now

closed 2017-02-09 by NickDraper