mantidproject/mantid #16167: Unscripted Testing Documentation

NickDraper, 2016-05-12:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this ticket once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

Qt Assistant offline documentation
* Algorithm, fit, concept and api pages should be generated
* Algorithm dialog snapshots should appear on algorithm pages in offline help
* Math formulae should appear on algorithm pages in offline help
* workflow diagrams should appear on algorithm pages in offline help

NickDraper, 2016-05-16:
Testing on Windows 7

NickDraper, 2016-05-16:
Tested offline docs, concepts, algs functions and api.

Images, math formulae, workflow diagrams and interface images.

DavidFair, 2016-09-20:
There are a couple of lines of text and links which are failing to be output due to syntax errors. Currently going through and fixing them up.

DavidFair, 2016-09-20:
Tested against Windows 10 - 
Images, formulae, diagrams and interfaces images display correctly. 
Remaining issues should be fixed with #17541

raquelalvarezbanos, 2017-01-26:
Everything looks good (images, equations, diagrams). Tested on Windows.

closed 2017-01-26 by raquelalvarezbanos