mantidproject/mantid #16169: Unscripted Testing ISIS Reflectometry

NickDraper, 2016-05-12:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this ticket once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

For testing procedures see http://www.mantidproject.org/Testing_ISIS_Reflectometry_GUI

MattKing06, 2016-05-16:
After completed unscripted testing for the old Reflectometry interface, any issues found will be cataloged [here](https://github.com/mantidproject/mantid/issues/16227)

raquelalvarezbanos, 2016-09-20:
While testing the interface I found a bug when plotting (https://github.com/mantidproject/mantid/issues/16169). I am not going to have time to have a look at it today (I am going to be in a course), so I am leaving it unassigned.

raquelalvarezbanos, 2016-09-21:
Both interfaces seem to be working OK, I haven't found any issues.

raquelalvarezbanos, 2017-01-25:
Everything seems OK. Both interfaces are working as I would expect.

closed 2017-01-25 by raquelalvarezbanos