mantidproject/mantid #16170: Unscripted Testing Mantidplot Group 4

NickDraper, 2016-05-12:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this ticket once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

tom-perkins, 2016-05-17:
Tested 16/5 nightly build on Windows 7

| Area to test | Result |
| --------------- | --------- |
| About MantidPlot | |
| Qt help | |
| Data loading | |
| 2D graphs | |
| 3D graphs | |
| Matrices | |
| Graph axes, labels | |
| Config/Properties | |
| ICAT | |
| Show detectors | |
| Sample Logs | |
| Numpy | |
| Scripts | |
| Python error handling | |
|

tom-perkins, 2016-09-21:
Testing Mantid 3.7.20160920.1910 on Ubuntu 14.04

| Area to test | Result |
| --------------- | --------- |
| About MantidPlot | ✓ |
| Qt help |  |
| Data loading |  |
| 2D graphs |  |
| 3D graphs |  |
| Matrices |  |
| Graph axes, labels |  |
| Config/Properties |  |
| ICAT |  |
| Show detectors |  |
| Sample Logs |  |
| Numpy |  |
| Scripts |  |
| Python error handling |  |

AntonPiccardoSelg, 2017-01-24:
Testing Mantid Nightly Release 24/01/2017 on Windows 7

| Area to test | Result |
| --- | --- |
| About MantidPlot | ✓ |
| Qt help | ✓ |
| Data loading | ✓|
| 2D Graphs|

closed 2017-01-24 by AntonPiccardoSelg