mantidproject/mantid #16171: Unscripted Testing Mantidplot Group 2

NickDraper, 2016-05-12:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this ticket once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

tom-perkins, 2016-09-21:
Testing Mantid 3.7.20160921.821 on Windows 7

| Area to test | Result |
| --------------- | --------- |
| About MantidPlot | ✓ |
| Qt Help | |
| Data loading | |
| 1D graphs | |
| QtiPlot graphs | |
| Tables | |
| Graph toolbars | |
| Instrument view | |
| Help | |
| IPython | |
| QtiPlot fitting | |
| Mantid fitting | |
| Mantid sequential fitting | |
| Mantid user functions | |

AnthonyLim23, 2017-01-24:
About Mantidplot: Y
QTHelp: Y
Load data: Y (loaded a few files)
Tiled windows: Y (added and removed plots)
tables: Y (could be generated and loaded) 
graph toolbar buttons: Y (could draw on plots and add functions) 
Instrument view: Y (checked each view. Masking and labelling worked). 
Help (menu and algorithm): Y (clicked a help on a few random algorithms and tried all of the options in the help menu)
ipython: Y (worked for a simple notebook). 
MantidFitting: Y (fitted a few functions) 
 sequential fitting: Y (I did get the parameters as a function of log, but some where NAN and it did exit gracefully). 
user function: Y (fitted a parabola via user defined function)

closed 2017-01-25 by AnthonyLim23