mantidproject/mantid #16172: Unscripted Testing Muon

NickDraper, 2016-05-12:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this ticket once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

As a minimum test http://www.mantidproject.org/Unscripted_Manual_Testing_MuonAnalysis.
For more info see http://www.mantidproject.org/Muon

tom-perkins, 2016-05-16:
Windows 7 (as used by majority of muon scientists at present)

Note there is currently a crash using the muon interfaces on Ubuntu (#15984) - currently investigating.

tom-perkins, 2016-05-16:
Tested by running through the Mantid muon manual and the training school workbook. Found bugs #16224 and #15984.

tom-perkins, 2016-09-19:
Testing on Windows 7 using the muon manual and 2014 training school workbook.

tom-perkins, 2016-09-22:
Re-testing on Windows 7 because ~~#16855~~ is merged and the interface has significantly changed.
Using the new unscripted testing guide.
Also will test the ALC interface, which I didn't test before.
- [ ] Muon Analysis
- [ ] ALC

KarlPalmen, 2017-01-25:
I'm testing in Windows 7.

KarlPalmen, 2017-01-25:
I'm testing the release of January 23.

KarlPalmen, 2017-01-25:
All the scripted tests went OK, except test 7, which could not be completed, because the `Label for Simultaneous Fit`  box was disabled.

KarlPalmen, 2017-01-25:
This was because I followed the instructions in the order given. If follow the next instruction (set period checkboxes) first, I can then set the label.

KarlPalmen, 2017-01-25:
When I change the function to Linear Background (in group 2 test 7), I get an unexpected exception `ParamFunction LinearBackground does not have parameter (f1.Phi)`.

KarlPalmen, 2017-01-25:
I have not been able to reproduce this unexpected exception in a simple manner. It may have occurred because of a change in instrument. 

I've found another exception that occurs when one changes instrument. This can be produced by selecting an instrument do some data analysis on it then return to the home tab, then change instrument, then observe that the data analysis tab is still enabled, even though no data has been loaded for newly selected instrument. Selecting the data analysis tab leads to an unexpected exception.

KarlPalmen, 2017-01-25:
Group 3 (Results tables) not tested owing to above-mentioned unexpected exception in group 2 test 7.

KarlPalmen, 2017-01-26:
I tried group 2 test 1 followed by group 2 test 7 and when I tried to fix `linear background` and exception came up complaining about a non-existing parameter. I'll raise this as an issue.

KarlPalmen, 2017-01-26:
I tested group 3 using the Mantid overnight build of 25 January having run all of group 2, except test 7 and found the result tables were correct, taking account of the absence of the MUSR results arising from test 7 not being done.

KarlPalmen, 2017-01-26:
Everything OK except issue #18566.

closed 2017-01-26 by KarlPalmen