mantidproject/mantid #16173: Unscripted Testing Mantidplot Group 3

NickDraper, 2016-05-12:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this ticket once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

AntonPiccardoSelg, 2016-05-17:
Testing on Windows 7:

| Functional area                          |  Result |  
| ----------------------------------------| ----------|  
| About MantidPlot (has all icons) | Icons are there |
| Qt Help Documention                 | Documentation seems to be fine for algorithms, concepts, etc.|
| Data Loading  | Loading .nxs (event and histo) and .raw is fine |
| Facility catalog searching and downloading | Can log in, search, download and load|
| 2D graphs | Played with contour plot, axes, styles --> seems fine |
| Qtiplot | Slice Viewer, Spectrum Viewer seem to display correctly|
| Matrices | Seems to behave fine; Image mode has black block |
| Printing | Does not seem to work |
| Graph toolbar buttons | Played with all options, seems to be fine |
| Workspace Send to  | Where is this???????|
| IPython interpreter | Open Ipython interpreter with shortcut and created sample workspace, seems fine|
| Python Workspace Algebra | Plus, Multiply, Divide, Subtract(event catches unit difference); seems fine |
| Running scripts | Executed some scripts of mine, seems fine |
| Algorithm execution | Executed severeal algorithms, seems fine |
| Algorithm Abortion | Aborted loadin of large MD workpspace, worked fine|

AntonPiccardoSelg, 2016-05-17:
Items to further investigate:
* Workspace Send to
* Printing
* Matrices (View mode) --> should be a ticket

DavidFair, 2016-09-21:
Windows 10

- Matrices still rendering out with a black section #16246
- "Sent to..." I couldn't find it even with a MomentumTransfer workspace. Issue open in #11571
- `LoadEventNexus` leaks memory on cancel - possibly other loaders do this too but I don't have large enough files to replicate it. #17565

Everything else works correctly

AnthonyLim23, 2017-01-25:
Testing on Windows 7:

| Functional area | Result |
| --- | --- |
| About MantidPlot (has all icons) | Icons are there |
| Qt Help Documention | Documentation seems to be fine, except for two things. Concepts has two G's and two P's in its contents. Algorithm help has two F's and two P's in its contents page. |
| Data Loading | Loaded a few files, the progress bar for RAW files seems to hang at 98 or 99%, however the output does confirm that the data has been loaded. |
| Facility catalog searching and downloading | Can search and load data. |
| 2D graphs | Made a color fill plot and edited it (added contours, changed the scale). |
| Qtiplot | Spectrum Viewer appeared to work.|
| Matrices | The black box still appears in image mode: https://github.com/mantidproject/mantid/issues/16246 |
| Printing | Seems to work. |
| Graph toolbar buttons | Seems to work. |
| Workspace Send to | To set up send to need to go into preferences and add a file. After that a 'send to' option appears in the ws menu. However, when I try to create the send to it keeps the save icon blacked out if I use the hdfview executable but allows me to save if I use the install wizard.  |
| IPython interpreter | The python script window seems to work. I ran some code form the exercises. |
| Python Workspace Algebra | Plus, Multiply, Divide, Subtract, seems fine |
| Running scripts | Executed some scripts of mine, seems fine |
| Algorithm execution | Seems fine |
| Algorithm Abortion | seems fine |

AnthonyLim23, 2017-01-25:
I have done some more testing and the hang only occurs after attempting to load data from the catalog. It is irrespective of the file type and all subsequent loads will also hang. https://github.com/mantidproject/mantid/issues/18544

closed 2017-01-25 by AnthonyLim23