mantidproject/mantid #16174: Unscripted Testing TofConverter

NickDraper, 2016-05-12:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this ticket once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

NickDraper, 2016-05-13:
Testing on Windows 7

NickDraper, 2016-05-13:
Simple, but works

raquelalvarezbanos, 2016-09-21:
Works fine (Windows 7, Mantid version 3.7.20160919.1922).

AnthonyLim23, 2017-01-25:
Windows 7, it does not convert from x to Nu(Thz) and produces the pop-up box 'local variable OutputVal referenced before assignment. However, reversing the conversion yields results. See https://github.com/mantidproject/mantid/issues/18553

closed 2017-01-25 by AnthonyLim23