mantidproject/mantid #16176: Unscripted Testing QECoverage

NickDraper, 2016-05-12:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this ticket once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

AndreiSavici, 2016-05-18:
Works as documented. Tried different instruments, checked for consistency with ConvertToMD. Using bad input values yield nice error messages, and the interface does not crash. Documentation is OK.
OS used Ubuntu 15.10

mducle, 2016-09-23:
Tested on Windows 7 and RHEL 7 with nightly build download 20160922.1557 and self-compiled from `release-v3.8` branch. 

Works as documented. Bad inputs gives error messages and no crashes.

AndreiSavici, 2017-02-07:
Works as documented on Ubuntu 16.10. No crashes for bad input

closed 2017-02-07 by AndreiSavici