mantidproject/mantid #16179: Unscripted Testing Multi dataset fitting

NickDraper, 2016-05-12:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this ticket once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

AntonPiccardoSelg, 2016-05-17:
Following http://www.mantidproject.org/Fitting_QENS_I%28Q,_t%29

Some issues:
* Workspace index seems to be inverted
* Cannot clear old model (at least not the guess, etc which are plotted)

tom-perkins, 2016-09-21:
1. Following [instructions](http://www.mantidproject.org/Fitting_QENS_I%28Q,_t%29) with `irs26176_graphite002_iqt` in sequential fit.
  - Had to turn on "Create Output" for it to work, this should probably be selected by default. If it is not selected, no output is plotted and no parameters workspace is generated.
  - Got warnings "StretchExp is undefined for negative argument" for a fitting range that was all in positive x. Fit worked OK, though.

2. Use case from muon scientists: Asymmetry from EMU20918-27 with Abragam function. Simultaneous fit with all parameters global except Tau.
  - OK: this time no need to select "Create Output" for simultaneous fit

AntonPiccardoSelg, 2017-01-25:
Seems as before. Old issues are still there, but is nevertheless fine to use.

closed 2017-01-25 by AntonPiccardoSelg