mantidproject/mantid #16180: Unscripted Testing FilterEvents

NickDraper, 2016-05-12:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this ticket once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

NickDraper, 2016-05-20:
testing on windows 7

NickDraper, 2016-05-20:
Works, but the usage of the dialog is far from intuitive

DavidFair, 2016-09-20:
Tested on Windows 10
Interface works and correctly filters and outputs. It also handles bad input correctly. 
My only minor quibble is that it says `File/Run:` next to a field which only takes files whilst runs are selected in the drop down below that field.

AnthonyLim23, 2017-01-25:
I have loaded an appropriate file. Messages are presented if the file is incompatible. 

It is possible to put letters into the Time Interval of the Filter By Time tab (other boxes that expect numerical values do not allow non-numeric values). If a filter is then applies the error 'could not convert string to float' is given in results log. 

All of the other options do run and the output is different (I am not checking that the output is correct). 

For some Sample logs the line (on the plot) goes backwards in time (e.g. CNCS_7860_event 
[CNCS_7860_event.zip](https://github.com/mantidproject/mantid/files/729960/CNCS_7860_event.zip) with sample logs of 'Speed4' and Phase1).

AnthonyLim23, 2017-01-25:
Everything else works as expected

closed 2017-01-25 by AnthonyLim23