mantidproject/mantid #16182: Unscripted Testing DGSPlanner

NickDraper, 2016-05-12:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this ticket once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

AndreiSavici, 2016-05-18:
Works as expected. Tested for consistency.
The help button is not working (pointing to some missing page). The fix is in PR #16290

lottiegreenwood, 2016-09-26:
Works as described in http://docs.mantidproject.org/v3.7.1/interfaces/DGSPlanner.html

AnthonyLim23, 2017-01-26:
Everything works as expected, except I cannot load a mask (I do not currently have a  compatible file)

AnthonyLim23, 2017-01-26:
Generated a mask from instrument view and selecting 'Apply to data' and  then saving the workspace as a nexus file.  When loaded into the interface it behaved as expected.

closed 2017-01-26 by AnthonyLim23