mantidproject/mantid #16184: Unscripted Testing Vates

NickDraper, 2016-05-12:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this ticket once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

Including data loading in Vates via ParaView. For help see [[Testing_Notes_2.5.0|here]]

NickDraper, 2016-05-19:
Tested by Owen

lottiegreenwood, 2016-09-23:
windows 7

lottiegreenwood, 2016-09-23:
tested using instructions and http://www.mantidproject.org/VatesSimpleInterface_v2
found https://github.com/mantidproject/mantid/issues/17582 but no major issues

lottiegreenwood, 2017-01-24:
windows 7

lottiegreenwood, 2017-01-24:
tested using instructions, confirm https://github.com/mantidproject/mantid/issues/17582 still exists but otherwise no major issues

closed 2017-01-24 by lottiegreenwood