mantidproject/mantid #16185: Unscripted Testing Engineering Diffraction

NickDraper, 2016-05-12:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this ticket once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

FedeMPouzols, 2016-05-17:
This interface was demoed last week with instrument scientists which was effectively an early beta testing session.
@shahrozahmed1 and me are working on it these days, and during the beta testing period we'll be fixing some usability issues. Some of them are being tested right now: #16229, #15991.

FedeMPouzols, 2016-05-17:
The issues opened against the "Engg GUI"  can be seen here:
https://github.com/mantidproject/mantid/issues?q=is%3Aopen+is%3Aissue+label%3A%22Component%3A+Diffraction%22+label%3A%22Component%3A+GUI%22
(with a few obviously unrelated issues that have to do with different diffraction GUIs).

FedeMPouzols, 2016-09-16:
For the tester, we now have a testing guide for this interface: http://docs.mantidproject.org/nightly/interfaces/Engineering_Diffraction_TestGuide.html

DimitarTasev, 2016-09-21:
Testing on Windows 7, [build 3.7.2.20160919.1922-win64](http://builds.mantidproject.org/job/master_clean-win7/290/)

DimitarTasev, 2016-09-21:
Everything looks good

gemmaguest, 2017-01-30:
I found the following issues when testing this: #18607, #18609, #18611, #18612. 

Most of these problems have been there since at least version 3.8.0 and are probably not urgent, although #18607 is a possible regression.

closed 2017-02-08 by gemmaguest