mantidproject/mantid #16186: Unscripted Testing Step scan analysis

NickDraper, 2016-05-12:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this ticket once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

See http://www.mantidproject.org/Step_Scan_Interface

AndreiSavici, 2016-05-19:
Tested with HYS111669. Plotted against `omega` and the `...Axis2` (the same log with a different name), for different normalization options.
Tested masking some peaks.
Everything seems to be working fine

DimitarTasev, 2016-09-21:
Tested with HES111669 on Windows 7, [build 3.7.2.20160919.1922-win64](http://builds.mantidproject.org/job/master_clean-win7/290/)

Everything seems to be working fine

raquelalvarezbanos, 2017-02-09:
Tested on Windows 7, seemed to work fine.

closed 2017-02-09 by raquelalvarezbanos