mantidproject/mantid #16187: Unscripted Testing Workspace Groups

NickDraper, 2016-05-12:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this ticket once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

Ensure MantidPlot displays Group Workspaces correctly in the workspace tree

NickDraper, 2016-05-17:
windows 7

NickDraper, 2016-05-17:
This works really nicely, the group plotting features are very good.

FedeMPouzols, 2016-05-17:
Note that, not about plotting but manipulation of workspace groups in general, there are a number of glitches  that sometimes cause crashes or bad behavior with algorithms: #10709,  #11361, #14249, #8652, and more in an old comment here: http://trac.mantidproject.org/mantid/ticket/6408.

DavidFair, 2016-09-20:
- #8652 Is still present as this script will produce two workspaces with the same name.
```python 
CreateSampleWorkspace(OutputWorkspace='bob', Function='User Defined', UserDefinedFunction='name=LinearBackground,A0=3.22542,A1=-0.0633193')
CreateSampleWorkspace(OutputWorkspace='bob1', Function='User Defined', UserDefinedFunction='name=LinearBackground,A0=3.22542,A1=-0.0633193')
CreateSampleWorkspace(OutputWorkspace='bob2', Function='User Defined', UserDefinedFunction='name=LinearBackground,A0=3.22542,A1=-0.0633193')
CreateSampleWorkspace(OutputWorkspace='bob3', Function='User Defined', UserDefinedFunction='name=LinearBackground,A0=3.22542,A1=-0.0633193')
GroupWorkspaces(InputWorkspaces='bob,bob1,bob2,bob3', OutputWorkspace='NewGroup')
RenameWorkspace(InputWorkspace='bob1', OutputWorkspace='bob')
```

- Attempting to `Plot a contour` from a group will result in the exception `Xmode is not BinEdges`

DavidFair, 2016-09-21:
Following me previous comment we have opened #17551 and postponed it to release 3.9 as the incorrect behaviour is also present in the current release 3.7.2.

Everything else works well.
Windows 10.

AntonPiccardoSelg, 2017-01-24:
GroupWorkspaces seem to be fine. I found one item which is worth investigating #18534
Tested Nightly Release from 23/01/2007 on Windows 7

closed 2017-01-24 by AntonPiccardoSelg