mantidproject/mantid #16188: Unscripted Testing ORNL Powder Diffraction

NickDraper, 2016-05-12:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this ticket once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

See http://www.mantidproject.org/PowderDiffractionReduction

peterfpeterson, 2016-05-25:
Ashfia found #16399.

NickDraper, 2016-05-31:
Closed at the end of Beta Testing

closed 2017-07-03 by NickDraper