mantidproject/mantid #16189: Unscripted Testing Sample Transmission calculator

NickDraper, 2016-05-12:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this ticket once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

NickDraper, 2016-05-17:
windows 7

Tested valid and incorrect input
both single and multiple modes
linear and logarithmic binning
help & documentation

raquelalvarezbanos, 2016-09-21:
Tested on Windows 7 (Mantid version 3.7.20160919.1933).

Everything seems OK. I tried to break the interface by setting wrong formulas, invalid rebinning parameters, etc.

soininen, 2017-01-25:
Tested on Ubuntu 14.04.

There were two issues: #18546 and #18552 
Run the tests on a version where the first issue was fixed (PR #18549). Tried both valid and invalid inputs, single and multiple modes, regular and logarithmic binning. After the locale issue fix, everything seems to work OK.

closed 2017-01-25 by soininen