mantidproject/mantid #16191: Unscripted Testing Mantidplot Group 1

NickDraper, 2016-05-12:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this ticket once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

NickDraper, 2016-05-16:
Testing on Windows 7

NickDraper, 2016-05-16:
project loading notes added to #15602

NickDraper, 2016-05-16:
All other areas ok

DimitarTasev, 2016-09-20:
Testing on Windows 7, [build 3.7.2.20160919.1922-win64](http://builds.mantidproject.org/job/master_clean-win7/290/)

DimitarTasev, 2016-09-21:
Multiple issues found, tracking issue: #17548

DimitarTasev, 2016-09-22:
All issues in #17548 have been resolved. Ready to be closed

AntonPiccardoSelg, 2017-01-25:
Testing on Windows 7:

| Functional area | Result |
| --- | --- |
| About MantidPlot (has all icons) |Ok |
| Qt Help Documention  |Ok |
| Data Loading | Ok |
| Facility catalog searching and downloading | Cannot use the data search archive; See |
| Load/Save Projects | Ok|
|1D Graphs (including waterfall and tiled plot options)| Ok|
|3D graphs | Ok|
|Tables |  Ok|
| Printing | Ok|
| Graph Manipulation - axes / labels etc| Ok|
|Catalog (icat) searching & downloading| see #18529|
|Workspace Save/Load to Nexus | Ok|
| Workspace History | Ok |
| Script Window  | Ok |
|Algorithm execution | Ok|
| Algorithm Custom Dialogs| Ok |

closed 2017-01-25 by AntonPiccardoSelg