mantidproject/mantid #16195: EnggGUI: preIntegFile and PreCurveFIle files not saved after EnggVaandiumCorrection

shahrozahmed1, 2016-05-13:
### Expected behavior
When new calibration is carried out, if the file is found; the `EnggVanadiumCorrection` are calculated first which produces `preIntegFile` and `preCurveFile`. These files should be saved so next time `EnggVanadiumCorrection` wont have to be calculated for the same `vanadium` and `ceria` run number and existing `preIntegFile` and `preCurveFile` can simply be loaded, which would save sufficient time and speed up the calibration process.
### Actual behavior
Currently `EnggVanadiumCorrection` are being calculated every time the calibration is carried out as the `preIntegFile`. 
### Steps to reproduce the behavior
Run calibraiton on `EngineeringDiffraction` GUI
### Platforms affected
All

FedeMPouzols, 2016-05-13:
This is more a missing feature than a bug. The interface is smart enough to reload precalculated results but not enough to save them for new vanadium runs. So let's give it lower priority than the other issues, especially for the beta testing period where we should minimize changes and avoid risks.

NickDraper, 2016-05-20:
If this is low priority should we move it to R3.8?

FedeMPouzols, 2016-05-20:
Yes, definitely.

shahrozahmed1, 2016-06-17:
- [ ] Prerequisite #16649

DavidFair, 2016-09-12:
I need to check if #17424 is a duplicate of this or this is a separate issue.

FedeMPouzols, 2016-09-16:
Yes, #17424 and this issue are the same.

closed 2017-01-16 by NickDraper