mantidproject/mantid #16199: SCD calibration issues

VickieLynch, 2016-05-13:
Fix issues found by instrument scientists testing SCDCalibratePanels

- Lattice should also be optimized and match input to 3-4 decimal places

- Fix width, height, and t0 on first pass (t0 and l0 are correlated)

- Check how depth and detd are used and if they are correct after optimization

- Slow (4-5 minutes) for MANDI 

- XML output not correct from parallel region

NickDraper, 2016-05-20:
Vickie,  Is this correctly aimed at R3.7?  Do all of the issues above need to be in that release?

NickDraper, 2016-05-31:
Vickie can you give me an update on this or at least the parts that need to get into R3.7?

NickDraper, 2016-06-02:
Any changes that need to get into release 3.7 need to be in a tested PR by the end of today.

NickDraper, 2016-06-03:
Moved to R3.8 for the remaining aspects of this issue

PR #17153 (2016-08-19): Rewrite of SCDCalibratePanels
Description of work.
SCDCalibratePanels   has been rewritten to calibrate the position and rotations of each panel independently in parallel.  There are options to calibrate the panel size and the L1 for all the panels.
 Only the U of the UB matrix is refined.  There is a script SCDCalibratePanelsResults.py that takes the output of  this algorithm and plots the theoretical vs calculated position of each peak for each panel.  The RMSD in mm is  calculated and printed.

**To test:**
https://www.dropbox.com/s/5eidrhs9ehnfkge/Si2mm_Cubic_F.integrate?dl=0
```python
Loa [... cut]
closed 2016-08-19 by quantumsteve