mantidproject/mantid #16244: Make it possible to load the information contained in the 'Options' column

raquelalvarezbanos, 2016-05-17:
Modify `SaveTBL` and `LoadTBL` to make it possible to load the information contained in the 'Options' column in the Reflectometry interface.

raquelalvarezbanos, 2016-05-17:
We should be able to deal with complex options:

`AnalysisMode="PointDetectorAnalysis",ProcessingInstructions="1,2,3",CorrectDetectorPositions=1`

MattKing06, 2016-06-30:
The ISIS Reflectometry (Polref) interface is an example of a GenericDataProcessor. Previous to the creation of the GenericDataProcessor, we would load ReflTBL files into the interface by using an algorithm called `LoadReflTBL()`. Since the GenericDataProcessor now needs to be used for multiple technique areas, we need a more general version of the ReflTBL that can have column-headings associated with the technique area (like Qmin and Qmax columns are associated to Reflectometry).

We have generalized `SaveTBL` and `LoadTBL` algorithms so that they can now have column headings specified the GenericDataProcessor in which they are being used. However, each TBL must have a "Group" column of type Integer and "Options" column that is of type string. When loading and saving the TBL file we found that there needs to be extra work put into how the "Options" column is parsed with respected to commas.

Currently we cannot handle complex strings that contain Speech-marks and multiple commas such as: AnalysisMode="PointDetectorAnalysis",ProcessingInstructions="1,2,3",CorrectDetectorPositions=1.

You can find LoadTBL and SaveTBL cpp files at:
`...\mantid\Framework\DataHandling\src`

gemmaguest, 2019-07-25:
This will be done by #18826

closed 2019-07-25 by gemmaguest