mantidproject/mantid #16253: 1D Graphs: "Fill Area" not filling area properly

raquelalvarezbanos, 2016-05-17:
Issue discovered during unscripted testing.

Load a dataset and plot some spectra using the "Waterfall Plot" option (for instance, I tried with "EMU00020711"). Click on "Reverse Order", then "Fill Area" and check "Enable Fill", areas under the curves are not displayed properly.

raquelalvarezbanos, 2016-07-12:
[EMU00020711.zip](https://github.com/mantidproject/mantid/files/359185/EMU00020711.zip)

DimitarTasev, 2016-07-13:
Could not replicate the issue. I have also tried with different data files, and still could not replicate. Following are the images of my result:

Loading "EMU00020711.nxs" and plotting spectra range 1-3 using "Waterfall Plot"
[image]

"Reverse Order" has been clicked and you can see that the plots are reversed
[image]

"Fill Area" has been clicked and "Enable Fill" has been checked, the filled area looks normal
[image]

DimitarTasev, 2017-01-17:
Is this still a valid issue? I could not replicate the issue, but it wasn't closed after being reviewed?

NickDraper, 2017-06-29:
This cannot be reproduced anymore

closed 2017-06-29 by NickDraper