mantidproject/mantid #16267: Change tests of `CurveFitting` "functions" to be actual unit tests

NickDraper, 2016-05-17:
### Expected behavior
We want to have unit tests that just test the unit aspect of the code. Including boundary conditions etc.

Complex tests of the complete fitting framework should move out tu the Fit unit testing or a system test.

### Actual behavior
Currently the unit tests for many of the functions test much more than the function itself, often performing optimised fits.

FedeMPouzols, 2016-05-25:
@raquelalvarezbanos

FedeMPouzols, 2016-05-25:
Dividing into blocks:

- Functions in the category "Muon": #16418.

raquelalvarezbanos, 2016-05-27:
Functions in the category "Peaks": https://github.com/mantidproject/mantid/issues/16452

FedeMPouzols, 2016-09-22:
@NickDraper: I just pushed this back to 3.9. This is 95% done, just the "QuasiElastic" functions need a bit of attention.

gemmaguest, 2020-10-29:
Closing as stale because we are unlikely to get around to this

closed 2020-10-29 by gemmaguest