mantidproject/mantid #16280: Context menu options for read-only tables

tom-perkins, 2016-05-18:
Noticed during a maintenance task in `ApplicationWindow.cpp`

In `showTableContextMenu`, there are some actions that are only added to the context menu if the table is editable.
This works in the form
```c++
if (isEditable)
    cm.addAction(action1);
    cm.addAction(action2); // this is added whether editable or not?
if (isEditable)
    ...
```
Some options are added to the context menu whether or not the table is editable. These include "delete rows", which seems like it might be a mistake. (You can't insert rows into a read-only table, or cut rows out, but you can delete rows...) This issue is to check whether the behaviour is as intended or not.

Example: [see here](https://github.com/mantidproject/mantid/blob/6514ee1d6a0ad14096af7ba236a1fe5a80434bf3/MantidPlot/src/ApplicationWindow.cpp#L10378-L10382)

Even if this is as intended, should add braces to the `if` statements to make it clearer which actions depend on `isEditable` and which do not.

NickDraper, 2016-06-07:
@tom-perkins Could you comment on why a review is required here?  Are you suggesting this is a duplicate, no longer valid, or just not worth fixing?

tom-perkins, 2016-06-13:
@NickDraper The 'review required' is to confirm whether the behaviour here is as intended (no bug) or whether it is a bug.

closed 2020-07-29 by gemmaguest