mantidproject/mantid #16288: Constraints and ties on same fit parameters

tom-perkins, 2016-05-18:
Question from a user via muon scientists: can you set constraints and ties on the same fit parameters?

Use case is something like:
[image]
The function is `FlatBackground + ExpDecayMuon + ExpDecayMuon`

Want to constrain each of `f0.A0`, `f1.A` and `f2.A` to positive values, but also tie them together so that their sum is a fixed number. This isn't possible with the FitPropertyBrowser - once you've set a tie, the only option is "Remove tie" and you can't add a constraint too.

It also doesn't seem possible when fitting with a script.
For example:
```python
from math import exp
import numpy as np

def func(x):
    return 0.1 + 0.01*exp(-0.02*x) + 0.015*exp(-0.015*x)
    
xData = []
yData = []
eData = []
for x in np.arange(0, 10, 0.1):
    xData.append(x)
    yData.append(func(x))
    eData.append(0.1)

ws = CreateWorkspace(xData, yData, eData, NSpec=1)

fitStatus, chiSq, covar, params, fitWS = Fit(InputWorkspace=ws, \
WorkspaceIndex=0, Output='fit', \
Function="name=FlatBackground,A0=0.01,constraints=(0<A0);name=ExpDecayMuon,A=0.2,Lambda=0.02,constraints=(0<A);name=ExpDecayMuon,A=0.02,Lambda=0.02,constraints=(0<A);ties=(f1.A=0.125 - (f0.A0 + f2.A),f2.A=0.125 - (f0.A0 + f1.A))")

print fitWS.getHistory()
```
and notice that the `Function` property of the `Fit` algorithm output is missing the ties (they've been lost).

NickDraper, 2016-06-07:
@tom-perkins Could you comment on why a review is required here?  Are you suggesting this is a duplicate, no longer valid, or just not worth fixing?

tom-perkins, 2016-06-13:
@NickDraper I didn't know whether setting constraints and ties on the same parameter is something that is supposed to be supported - i.e. whether the fact this doesn't work is a bug or by design. I also wanted to get the opinion of someone who is more familiar with this area of fitting.

NickDraper, 2016-06-22:
When you tie a parameter to  another one (or formulae) it is no longer optimised, as such a constraint would have no effect.

However if you tie fn2.p1 to fn1.p1, you can still put a constraint on fn1.p1 (as that is still optimised).

If that answers the question then please feed that back to the Muon scientists and close this issue.

tom-perkins, 2016-06-22:
Thanks, that answers the question. Closing the issue as "not a bug".

tom-perkins, 2016-06-27:
Reopening as there is a requirement to set up fits with parameters constrained like this, i.e:
f0.p1 + f1.p1 + f2.p1 = constant
f0.p1 > 0
f1.p1 > 0
f2.p1 > 0

and, at present, there is no way to do this in Mantid. The best you can do is set up the tie (f2.p1 = constant - f1.p1 - f0.p1) and constrain the other two parameters, but this still leaves the possibility that f2.p1 can be negative.